SRA-VJTI/MARIO #3: E (21596) esp-ros-wifi: Error receiving data: errno 113 E (22596) esp-ros-wifi: Error receiving data: errno 128

sergera-da-bahia, 2020-10-06:
Hi! 
I am trying to reproduce the project from the Workshop, but I'm facing some difficulties when flashing. It shows that the device is connected to the WiFi but I can't stablish any tcp connection.
If someone happens to know, have had the same issue before or have any thoughts on it, it would be very kind and I would be very thankful for it.
The errors from the ESP_LOGS, when monitor using (idf.py monitor) after flashing:

I (1586) esp-ros-wifi: Connected to AP
I (1596) esp-ros-wifi: Successfully connected
E (20596) esp-ros-wifi: Error receiving data: errno 113
E (21596) esp-ros-wifi: Error receiving data: errno 113
E (22596) esp-ros-wifi: Error receiving data: errno 128
E (23596) esp-ros-wifi: Error receiving data: errno 128
E (24596) esp-ros-wifi: Error receiving data: errno 128

The errors seem to be with the socket connection, but I am not any sure of it.
I am flasing the ESP32 using the ESP-IDF on a Windows 10 and ROS Melodic, instead of Linux and ROS Kinetic.
Thanks a lot in Advance to anyone who could have a look on it

VedantParanjape, 2020-10-06:
Hi,

I think there are three reasons for this issue:

1) rosserial node is not running on your PC

2) Windows Firewall is messing up with rosserial stuff

3) You have entered an incorrect ip address in menuconfig. 

It would be great if you could post your sdkconfig file here and a screenshot of the following command on windows PC, run it on cmd: `ipconfig/all`

Regards,
_**Vedant Paranjape**_

sergera-da-bahia, 2020-10-07:
Thank you very very much for the fast reply!
If it's in anyway not possible to open the zip files, here are my rosserial configurations in it and than the printscreen of my ipconfig /all

# # rosserial 
# CONFIG_ROSSERIAL_OVER_WIFI=y 
CONFIG_ROSSERVER_AP="TP-Link_9DB6" 
CONFIG_ROSSERVER_PASS="29233213" 
CONFIG_ROSSERVER_IP="192.168.0.138" 
CONFIG_ROSSERVER_PORT=11411 
# end of rosserial 
# end of Component config
[sdkconfig.zip](https://github.com/SRA-VJTI/ROS-Workshop-2.1/files/5338872/sdkconfig.zip)

[image]

VedantParanjape, 2020-10-07:
Hi,

Your Esp side config looks fine. Looking at the error codes, 

113 means connection abort: https://github.com/espressif/newlib-esp32/blob/4ddc99d47677b94c115cafcaa4a6939042ea8c4f/newlib/libc/include/sys/errno.h#L144

and 128 means socket is not connected which is justified since socket connection was aborted above: https://github.com/espressif/newlib-esp32/blob/4ddc99d47677b94c115cafcaa4a6939042ea8c4f/newlib/libc/include/sys/errno.h#L161

So, I can think of two reasons:

1) You are not running roscore and rosserial. Please attach a screenshot of the following command: `rosnode list`, to check if ros side is working correctly

2) Windows Firewall is blocking the connection, try again by disabling the firewall.

sergera-da-bahia, 2020-10-07:
Yes, that was one issue. I disabled the firewalls and my rosserial_python now finds something at least.
Then, the rosserial before that looked indefinately for a socket connection and never found one, thanks :)

But now I'm becomming those errors
on my ESP_LOGS:
E (7097) esp-ros-wifi: Error receiving data: errno 104
E (8097) esp-ros-wifi: Error receiving data: errno 128

and on my rosserial_python tcp closes automatically the socket:

[INFO] [1602063540.888584]: ROS Serial Python Node
[INFO] [1602063540.901593]: Fork_server is: False
[INFO] [1602063540.901593]: Waiting for socket connections on port 11411
[INFO] [1602063540.901593]: Waiting for socket connection
[INFO] [1602063549.240256]: Established a socket connection from 192.168.0.157 on port 54414
[INFO] [1602063549.255779]: calling startSerialClient
[INFO] [1602063551.362266]: Requesting topics...
[INFO] [1602063551.362266]: Client has exited, closing socket.
[INFO] [1602063551.362266]: Shutting down
[INFO] [1602063551.362266]: All done
Traceback (most recent call last):
  File "C:\Users\...\Documents\rosnoetic_catkin_ws\src\rosserial-noetic-devel\rosserial_python\nodes\serial_node.py", line 72, in <module>
    server.listen()
  File "C:\opt\ros\noetic\x64\lib\site-packages\rosserial_python\SerialClient.py", line 251, in listen
    self.startSerialClient()
  File "C:\opt\ros\noetic\x64\lib\site-packages\rosserial_python\SerialClient.py", line 257, in startSerialClient
    client.run()
  File "C:\opt\ros\noetic\x64\lib\site-packages\rosserial_python\SerialClient.py", line 466, in run
    if self.port.inWaiting() < 1:
  File "C:\opt\ros\noetic\x64\lib\site-packages\rosserial_python\SerialClient.py", line 308, in inWaiting
    chunk = self.socket.recv(1, socket.MSG_DONTWAIT|socket.MSG_PEEK)
AttributeError: module 'socket' has no attribute 'MSG_DONTWAIT'

Since that my rosnode list could only find /rosout
C:\Windows\System32>rosnode list
/rosout

VedantParanjape, 2020-10-07:
104 means connection reset by peer.

I am not sure what causes this issue. Maybe try again, by closing everything and flashing esp32 again.

sergera-da-bahia, 2020-10-07:
Hi,
It worked! Thanks for all the help!
I've followed the error from the link you gave.
https://stackoverflow.com/questions/54071217/attributeerror-module-socket-has-no-attribute-msg-dontwait
And then went further into the topic.
One suggested to set the socket.MSG_DONTWAIT to 0.
In this case in SerialClient.py line 308, in the fuction inWaiting:
chunk = self.socket.recv(1, socket.MSG_DONTWAIT|socket.MSG_PEEK

set to 0:
chunk = self.socket.recv(1, 0|socket.MSG_PEEK)

I am not sure, why it worked. My knowledge in this subject is yet quite limited
But it seemed to be an issue from windows
Again thanks a lot for looking into it

VedantParanjape, 2020-10-07:
Great that it worked, leave a star if you liked it.

> I am not sure, why. My knowledge in this subject is yet quite limited
But it worked :)

It didn't work as, windows networking library doesn't implement MSG_DONTWAIT feature where as linux networking library does, so it's a bug which needs to be fixed by ros guys, the solution is more of a hack. It is better if you use ROS melodic or linux.

sergera-da-bahia, 2020-10-07:
Before the Issue be closed. If I may, I've had a quick question...
I have noticed that it worked perfectly, while the ESP is powered on the USB port from my PC
Whereas when I plug in the power socket on the wall, the rosserial_python hangs by trying to connect to the socket
[INFO] [1602063540.888584]: ROS Serial Python Node
[INFO] [1602063540.901593]: Fork_server is: False
[INFO] [1602063540.901593]: Waiting for socket connections on port 11411
[INFO] [1602063540.901593]: Waiting for socket connection
...
And connects then only, when I plug it back into the USB port from PC..
Is that normal? was it actually intended to work like that? Or am I missing some configuration?

VedantParanjape, 2020-10-07:
> And connects then only, when I plug it back into the USB port from PC..
> Is that normal? was it actually intended to work like that? Or am I missing some configuration?

Whats the power rating of USB Power brick ?

This happens if esp doesn't get enough power, it's wifi fails to work correctly. Try doing this by plugging esp32 in some other PC, which is close to your router.

sergera-da-bahia, 2020-10-08:
Yes that was it. I changed the cable to a newer one and now it worked perfectly!
Again thank you very much!

VedantParanjape, 2020-10-08:
Glad, this project helped you.

:))

sergera-da-bahia, 2020-10-08:
Found directly in Github actually

VedantParanjape, 2020-10-08:
Nice.

I'll close this issue now.

closed 2020-10-08 by VedantParanjape