SRA-VJTI/MARIO #26: Porting MARIO to ROS2 and esp-idf to 5.0

aPR0T0, 2023-03-07:
- As ROS1 has ended its support for the latest ubuntu versions and as industry is progressing we need to adapt to the changes and port as soon as possible to the ROS2 latest distribution i.e. HUMBLE.
- Same goes for the esp-idf v5.0

Tasks Need to be done
- Porting to ROS2 Humble
- Designing a new end effector (maybe not gripper this time)
- Porting to esp-idf v5.0

aPR0T0, 2024-02-08:
As for the simulator, I have assigned SYs the task for doing the some research and finding which can be a better simulator instead of gazebo. 
Options to be considered:
- WeBots
- Gazebo latest (Won't be difficult and most of the code is ready in [this fork](https://github.com/aPR0T0/MARIO/tree/local/dev/3_simulation_gazebo))
- MuJoCo (Overkill for the small task as of MARIO)

Other tasks of porting will be assigned next week.

VedantParanjape, 2024-02-08:
> > * As ROS1 has ended its support for the latest ubuntu versions and as industry is progressing we need to adapt to the changes and port as soon as possible to the ROS2 latest distribution i.e. HUMBLE.
> > 
> >     * Same goes for the esp-idf v5.1
> > 
> >     * As ROSSERIAL is not supported for ROS1, one must look into the [MICRO-ROS](https://micro.ros.org/docs/tutorials/programming_rcl_rclc/node/) tutorials for writing nodes for esp32 and porting the firmware to microros
> > 
> > 
> > Tasks Need to be done
> > 
> >     * Porting to ROS2 Humble
> > 
> >     * Designing a new end effector (maybe not gripper this time)
> > 
> >     * Porting to esp-idf v4.4
> > 
> >     * OPTIONAL- Making a teleop controller to control the bot with webservice
> 
> @MOLOCH-dev @VedantParanjape can we start the project through this issue?

Looks good to me, you can implement it and see if it works well and is bug free. Probably do the development in a separate branch.

MOLOCH-dev, 2024-02-08:
Lgtm, a first step would be testing micro-ros with esp32 on ROS2 Humble. Another aspect would be using gz sim Harmonic instead of gazebo classic, I recommend porting the existing URDF to an SDF model. Create separate issues for testing micro-ros/Humble and rviz+ gz sim visualization, mention them in this thread. Update this current issue with a task list.

aPR0T0, 2024-02-10:
Migration from gazebo classic to latest gazebo (harmonic) (#38)

aPR0T0, 2024-02-10:
Migration from rosserial to microros (#39 )

AryanNanda17, 2024-03-21:
Hello @aPR0T0, 
- For porting the  [1_chatter_listener](https://github.com/SRA-VJTI/MARIO/tree/master/1_chatter_listener) from ros1_noetic to ros2_humble, should I create a different issue and do pull request and refer that issue or the PR should refer to this issue directly. 
- And, in the talker and listener folder, is there any need to add some additional stuff, or only porting is needed.

aPR0T0, 2024-03-21:
> Hello @aPR0T0,
>
>     * For porting the  [1_chatter_listener](https://github.com/SRA-VJTI/MARIO/tree/master/1_chatter_listener) from ros1_noetic to ros2_humble, should I create a different issue and [...]

You can create a separate issue for that
For porting use CMake only for don't use setup.py

AryanNanda17, 2024-03-22:
Alright!