SRA-VJTI/MARIO #27: Changing the gripper design

ChinmayMundane, 2023-03-28:
The aim of this issue is to design a new gripper for MARIO.
Older version of MARIO manipulator had an issue with its gripping mechanism. This time, we are thinking of designing new gripper for this purpose. 
Currently, There are two designs that we have thought of :

1. Gripper using rack and pinion mechanism : 
In this mechanism, gripping is done using rack and pinion structure which is supported/actuated by gear linked with motor.
[image]

2.  Gripper using more support
This design is similar to the previous one except it has more support which allows it to rotate in a proper manner.

[image]

aPR0T0, 2023-03-28:
I Think first design is pretty crude so let's first make it feasible first and my suggestion would pe get the hardware in hand of both and then compare which will be better

Jamm02, 2023-03-28:
> 2. Gripper using more support
>    This design is similar to the previous one except it has more support which allows it to rotate in a proper manner.
> 
> [image]

This design is not economical, keeping the costs in mind the designs should include minimum number of parts.

marck3131, 2023-03-28:
The older version had issues with the loose end / the driven gear, you can redesign the same with additional support. 
The 1st one can only grab objects perpendicular / and as Alqama already mentioned that the sliding length is too short to grasp other objects.
For the 2nd one, not mainly the gripper will cost more, but because of such design , you might have to redesign the other joints too, which would then be cost effective altogether, but you can go ahead by making a few changes and having a prototype in hand and test.

aPR0T0, 2023-03-29:
> This design is not economical, keeping the costs in mind the designs should include minimum number of parts.

@Jamm02 I don't think cost will be very high for this design, it has small links so should be feasible

Jamm02, 2023-03-30:
> > This design is not economical, keeping the costs in mind the designs should include minimum number of parts.
> 
> @Jamm02 I don't think cost will be very high for this design, it has small links so should be feasible

Make a prototype and provide a cost breakdown for the same. 
Be ready with atleast two bot prototypes by 10th April.

ChinmayMundane, 2023-04-23:
https://user-images.githubusercontent.com/109454803/233845934-c9d04851-fe0d-416c-b893-c2478196c231.mp4

The issue has been resolved.

closed 2023-04-23 by ChinmayMundane