SRA-VJTI/MARIO #30: Update installer script to install ROS2 and esp-idf v4.4

advait-0, 2023-04-10:
The current installer script needs to be updated to install ROS2 "Humble Hawksbill" and esp-idf v4.4, taking into account all the dependencies.

aPR0T0, 2024-02-10:
@advait-0 as sra-board-components have already been migrated to esp-idf v5.1, so you can restate the motive of this issue accordingly

NachtSpyder04, 2024-02-28:
If possible, can we update the script in such a way that all installations should be done when script is executed only once.
Personally, I feel like running the same script twice for installation doesn't look good and can be confusing for juniors

aPR0T0, 2024-02-28:
Hello @NachtSpyder04 , thanks for your concern. As we are porting now to ros2, we don't need to setup mambaforge environment. And there won't be the issue of running the script twice.

AryanNanda17, 2024-03-21:
Hello @aPR0T0, 
Could you please assign this issue to me? 
To make sure we are on the same page, could you please answer these questions:-
- The installation script should setup ros2 humble desktop  by following [these](https://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html) instructions and gazebo and it should create a ros2_ws and clone the repo inside it, right?
- The bash script should be created only for Linux users or for both Linux and MacOS? If both, then could you please share the installation guide for macOS users.

AryanNanda17, 2024-03-22:
Parallels Desktop is paid, UTM is a better option.
Your answers are clear to me!
Please assign this issue to me.

aPR0T0, 2024-03-23:
Good Luck with task!

AryanNanda17, 2024-03-25:
@aPR0T0,
For Mac users, should we go with the docker setup?
UTM has so many issues with running the gazebo.

AryanNanda17, 2024-03-25:
I can make a Dockerfile with the requirements.

SuperChamp234, 2024-03-25:
@AryanNanda17 go ahead with a docker setup. Make sure to include MicroROS in it as well.

aPR0T0, 2024-03-25:
Important packages:
1. Ros2 control all packages
2. Gazebo ros2 control
3. Joint state publisher 
4. Joint state publisher gui
5. Gazebo classic
6. Controller manager

AryanNanda17, 2024-04-08:
Hello @aPR0T0, 
Could you please tell me if I am missing something in the installation script [here](https://gist.github.com/AryanNanda17/ecfee629cd5286447e1e9f2b6c33454d).

aPR0T0, 2024-04-08:
Put the microrosagent in the rosws only

And please add a PR for this, so that review can be in-detail